prusa3d/PrusaSlicer #9832: Silent Install Should Not Open Sample Objects or Web Browser

HabitualTinkerer, 2023-02-22:
### Description of the bug

There is no method to prevent the sample objects folder or web browser from opening at the end of a silent install.

### Project file & How to reproduce

1. Run `prusa3d_win_2_5_0.exe /VERYSILENT /SUPPRESSMSGBOXES /NORESTART`
2. Notice that a Windows Explorer and the default web browser both open at the end of installation.

The issue can be corrected by adding the `skipifsilent` flag to each `postinstall` item in the `[Run]` section of `install_script.iss`. I would have submitted a pull request, but I couldn't find the Windows installer on GitHub; please consider adding it.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-win64

### Operating system

Windows 11

### Printer model

mk3s+

HabitualTinkerer, 2023-02-22:
Updated lines 379 through 384 in `install_script.iss` with `skipifsilent` flag added.

```
[Run]
Filename: "https://www.prusa3d.cz/jak-zacit/"; Description: "{cm:OpenTutorials}"; Languages: "czech"; MinVersion: 0.0,5.0; Flags: shellexec postinstall nowait skipifsilent
Filename: "https://www.printables.com/"; Description: "Navštívit komunitní hub Printables.com"; Languages: "czech"; MinVersion: 0.0,5.0; Flags: shellexec postinstall nowait skipifsilent
Filename: "https://www.prusa3d.com/new-user-guides/"; Description: "{cm:OpenTutorials}"; Languages: "english french italian german spanish"; MinVersion: 0.0,5.0; Flags: shellexec postinstall nowait skipifsilent
Filename: "https://www.printables.com"; Description: "Visit Printables.com - community hub"; Languages: "english french italian german spanish"; MinVersion: 0.0,5.0; Flags: shellexec postinstall nowait skipifsilent
Filename: "{commondocs}\Prusa3D"; Description: "{cm:OpenSamples}"; Components: mk3objects mk2objects mk2mmobjects mk3mmu2objects mk25mmu2objects; Languages: "english french italian german spanish polish czech"; MinVersion: 0.0,5.0; Flags: shellexec postinstall nowait skipifsilent
```

lukasmatena, 2023-11-27:
This should be fixed as of 2.7.0. Closing.

closed 2023-11-27 by lukasmatena