prusa3d/PrusaSlicer #9834: Alpha 2,3,4 no longer works on MacOS

army11b1980, 2023-02-22:
### Description of the bug

I was running Alpha 2 and 3. Then Alpha 3 crashed on start up. Downloaded Alpha 4. Same issue. Re-downloaded Alpha 2 and same issue, crash on startup. Can no longer run any of them on MacOS Ventura 13.2.1 M1 chip.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Alpha 2, 3, 4

### Operating system

Mac Ventura 13.2.1

### Printer model

MK3S+

tyrscgm, 2023-02-23:
I’m having the same issue it crashes it gives me the menu bar but no window

lukasmatena, 2023-06-16:
@army11b1980 @tyrscgm Do you guys still have this issue?

army11b1980, 2023-06-16:
No. No longer having this issue.

lukasmatena, 2023-06-18:
@army11b1980 Thanks. Then I suppose we can close this.

closed 2023-06-18 by lukasmatena