prusa3d/PrusaSlicer #9835: Organic Support through another object

Glitch-94, 2023-02-22:
### Description of the bug

Using Paint on-support enforcers, the support generated ignored another part and generated through it.
[image]

### Project file & How to reproduce

[MainFrontP1.zip](https://github.com/prusa3d/PrusaSlicer/files/10806657/MainFrontP1.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Windows 10 22h2 build 19045.2604

### Printer model

Prusa i3Mk3s+

neophyl, 2023-02-22:
All supports and even other objects have always done this in PS.  Its expected operation.  It just more noticeable now that organic supports are located further from the object they are supporting.  Each object is sliced individually.  Place 2 over lapping objects and you will have a very bad day.  You as the user are expected to avoid this.

To avoid this the correct procedure is to merge the objects.  In this way PS treats them as one and then supports etc do not ignore them.  This has been reported many many times already, they are closed as they are duplicates.

For example the most recent closed one https://github.com/prusa3d/PrusaSlicer/issues/9723
or https://github.com/prusa3d/PrusaSlicer/issues/9514
https://github.com/prusa3d/PrusaSlicer/issues/8273
https://github.com/prusa3d/PrusaSlicer/issues/7697

get the idea ?

Glitch-94, 2023-02-22:
Yea, sorry, tried looking for past examples and did not see them! Appreciate the feedback and sorry to open a duplicate.

closed 2023-02-22 by Glitch-94