prusa3d/PrusaSlicer #9837: For color blind people, please provide a better usable color bar scheme for numeric values

stefan-am, 2023-02-22:
I am red-green color blind and have big difficulties in assigning the color at a spot on the model to a value on the color bar. 
It reduces the fun to use PrusaSlicer … 

According to the link below, 8 % of all men and 0.5 % women are affected by this. 
For example, for my eyes the volumetric flow could be anything between 5 and 7.6 in this picture:

[image]

I found the following interesting source about this topic: 

[https://www.technia.com/blog/how-to-choose-abaqus-colormaps/ ](url)

I personally found greyscale colormaps much easier to use (in other software like FEA simulation programs). 
Maybe also the “Cividis colormap” mentioned in the link above should be considered: 

[image]

An very helpful “luxury” feature would be to read the value below the cursor tip and display it near the cursor, as demonstrated in this picture: 

[image]

kubispe1, 2023-02-22:
I like your second tip about some kind of mouse probe. Could be a nice feature, but for sure not part of 2.6.0 :D

tommymonk, 2024-05-04:
I just wanted to chime in to say that I too struggle immensely with the legend, I have to use a camera app (Color Grab) on my phone just to distinguish the values

closed 2026-09-29 by prusaslicer-bot[bot]