prusa3d/PrusaSlicer #9838: Organic Support Outside Model

1John419, 2023-02-22:
### Description of the bug

Perhaps this is intentional; just strikes me as odd. It may be related to #9836. 

Slicing the attached project generates organic supports which do not appear to actually support the model. The location is logical, as there is bridging at the location. However, it appears the support do not actually support the intended area.

[prusaslicer.zip](https://github.com/prusa3d/PrusaSlicer/files/10807750/prusaslicer.zip)

### Project file & How to reproduce

1. Open drydoc.3mf
2. Slice now
3. Inspect generated supports (see prusa.png).

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

Windows 10 Pro 64-bit 19045.2364

### Printer model

Prusa Mini+

kubispe1, 2023-02-22:
It is a different issue. I think it is all about set **Max Branch Angle** ( Print Settings/Support material) in combination with **Support on build plate only**. 
There are some cases to try:
- When you the angle to 20° there is no chance to generate support on build plate == no supports
- Your settings 40° it generate something, but out of the model
- set to 50° and it is ok.

1John419, 2023-02-22:
OK, settings issue; not a bug. Still learning about organic supports, but they are excellent. Yes, changing branch angle to 50 yields expected results. And setting supports to Everywhere also yields expected results. Apologies; user error. Thank you for pointing me in the right direction!

closed 2023-02-22 by 1John419

kubispe1, 2023-02-22:
Dont worry. In fact, at the moment it works in this way. For a bigger angle, there could be a risk of easier support collapse...