prusa3d/PrusaSlicer #9841: Crash when moving the window to another monitor

AssiveMass, 2023-02-23:
### Description of the bug

Prusaslicer crashes when moving the window to another monitor.

### Project file & How to reproduce

I cannot reproduce this repeatedly, seems pretty random. This has happened a few times. The objects have been quite large if that has to do with anything, custom 300x300 bed. I had multiple instances of PS open, all the newest alpha.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Windows 11 Business

### Printer model

Custom and MK3s+

mpaw, 2023-02-26:
I'm having the same issue but on `MacOS Ventura 13.2.1` on a M1 Pro Macbook Pro 16". I see the person reporting this is on windows. I'm also using the latest alpha `2.6.0-alpha4` but this issue has been going on for multiple production releases too.

Here are the various situations I have seen this happen
* I unplug my two external monitors while Prusa slicer window is open on one of the monitors. The screen on the macbook refreshes and PrusaSlicer crashes
* Same as above but it doesn't crash when I unplug the monitors. But then when I plug the monitors back in it crashes
* Macbook while plugged in to both monitors goes to sleep. On wakeup the slicer crashes
* Moving the PrusaSlicer window between the monitors causes a crash as soon as the window enters the viewport of another monitor

I have seen this happen when I have one or multiple PrusaSlicer instances open. Although from my experience when there are multiple open, not all of them crash.

Here is the crash log from when it crashed today after the MacBook (with both monitors plugged in) woke up from sleep.

[PrusaSlicer-Crash.log](https://github.com/prusa3d/PrusaSlicer/files/10834625/PrusaSlicer-Crash.log)

bubmann, 2023-04-11:
same here, Windows 10

pcout, 2023-07-12:
Same here. 
Windows 11. 
Prusa Slicer 2.6.0.

closed 2026-09-29 by prusaslicer-bot[bot]