prusa3d/PrusaSlicer #9843: bug stl PrusaSlicer2.6.0

tabletpauloroberto, 2023-02-24:
### Description of the bug

when I try to export to stl it goes to the directory to be saved and simply does not save in stl . this is the bug. thanks

### Project file & How to reproduce

xxxxxxxxxxx

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

e

### Operating system

e

### Printer model

e

kubispe1, 2023-02-24:
This is not a sufficient description of the bug. Please add some information. Export to STL works in 2.6.0-alfa4 ok.

- So attache 3mf
- attache video or gif 
Thanks

spineytoad, 2023-03-22:
[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/11042119/Test.zip)
I believe this is the same issue:
When I want to save this file with supports as .stl (file->export->export plate as stl/obj including supports), it doesn't save anything, it does save when the same file is saved and loaded into prusaslicer 2.5

kubispe1, 2023-03-23:
I used your .3mf and tested various combination of export
- common export
- no extension
- big capital letter for .STL
- And some others
I tried 2.5.0 and 2.5.1 and I didnot reproduce this issue.

closed 2026-09-29 by prusaslicer-bot[bot]