prusa3d/PrusaSlicer #9844: Interpunkčné znamienka v gcode

dushanL, 2023-02-24:
### Description of the bug

Pri zlúčení objektov je názov objektu pomenovaný v CZ verzii "Sloučení" bez možnosti premenovania. Ak mám zapnuté Označovanie objektov, tento názov je následne propagovaný do gcode s čím majú niektoré tlačiarne problém. V mojom prípade Klipper. Tento problém je aj pri jednotlivých objektoch, ale tie sa dajú premenovať.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Ubuntu 22.10

### Printer model

Klipper

kubispe1, 2023-02-27:
Hi, thanks for your issue. We made a small investigation... And hope that the last 2.6.0 alfa4 would work in your case. 
Can you try and confirm us if you can rename the merged object?

dushanL, 2023-02-27:
Hi, in version 2.6.0 alpha4, renaming the merged object is working.

closed 2023-02-27 by dushanL