prusa3d/PrusaSlicer #9846: Easier multi-material / multi-color print without MMU

smirgol, 2023-02-24:
It is still quite difficult / unintuitive to have a multi-color print where there are color changes *within* a layer, when doing manual filament changes.
There is, to my knowledge, no intuitive way to set this up and in the end you are required to manually check and edit the G-Code file to make it work properly.

First I add multiple extruders to my project. Then I used to add a M600 manually in the slicer to the Custom G-Code for "Color Change". This no longer works, as it is now considered a "Tool change". That's not necessarily intuitive - I'm changing the tool, because I need to change the color / material...

Then I need to edit the G-Code and:

1. Remove the first M600 right after printing the initial purge:
The filament is already loaded and purged, why unload it? There is no tool change at this point.
2. Modify/remove the initial purge on the wipe tower after filament change:
The filament is already fully loaded by the M600, no further purge is required to fill up the nozzle and it just results in a blob on the wipe tower, which will lead to serious issues over time (nozzle crashing at the blobs).
4. Remove all occurrences of "T0", "T1", "T2", ... as they seem to be MMU specific and just throw errors when not using an MMU.

Then there is this thing that might be firmware specific, which is that, after an M600, the nozzle moves to the last positions, drops filament and only then moves to the wipe tower. I think I've seen somewhere that this is fixed, probably need to update the firmware?

**Describe the solution you'd like**
I'd love to have a more straight-forward way to define a print as multi-material without an MMU. Please document settings in the extruder settings - it is not very clear what each option does: Do I need to pick "Single Extruder Multi Material?" It looks like you should, but I guess you shouldn't. It's not intuitive and you cannot read up on this on a website like you can for many other parameters.
It also would be nice if the printer would tell you which color / extruder number is expected at the current change  - right now I go to the sliced file and write down which color is required for which change.

**Describe how it would work**
Maybe some kind of a quick wizard, asking you about how many colors you'd like to use, if you're using an MMU or not and other things that might be relevant and then set up the slicer (and generated G-Code) accordingly.

**Describe alternatives you've considered**
There is none, you have to look at and edit the G-Code file and you need to know where to put which setting. If you don't, you are in for trouble. :)

ZzetT, 2023-09-10:
> Remove the first M600 right after printing the initial purge:
> The filament is already loaded and purged, why unload it? There is no tool change at this point.

I don't have a solution but other people suggest to add the following code to the toolchain g code:
`{if layer_num >= 0}M600{endif}`

> Then there is this thing that might be firmware specific, which is that, after an M600, the nozzle moves to the last positions, drops filament and only then moves to the wipe tower. I think I've seen somewhere that this is fixed, probably need to update the firmware?

Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2672

Hope this helps a bit.

epetit91, 2024-03-10:
The code `{if layer_num >= 0}M600{endif}` added to the toolchain g code works fine.

For the second problem (Duplicate of #2672), it is mentioned that this problem is fixed in version 2.7.2 of PrusaSlicer. I installed the latest version (on macOS 10.12.6) and the firmware 5.1.3+13503 of my MK4 and the problem is still present.

I haven't found any satisfactory solution to date, which is really frustrating.

Tupson444, 2026-09-18:
i guess this is still relevant, somebody should check before closing