prusa3d/PrusaSlicer #9849: Split External perimeters speed into two separate speed settings (inside/outside)

Misfi, 2023-02-24:
**Is your feature request related to a problem? Please describe.**
The inability to adjust the speed of external perimeters "inside" the model increases print times for some models significantly. 
One of the large containers I've printed required only ~**20 hours** to print after tricking the slicer to speed up the innermost external perimeters - versus the original ~**23 hours**. 
I believe the difference will be even more noticeable on large-volume printers like XL.
While it won't affect the majority of the models, it could increase the print speeds for ones with internal geometries that only have to be pretty on the outside - in my case the external perimeter had to be shiny and was printed slowly.

**Describe the solution you'd like**
The external perimeters speed setting would be replaced by two fields.
Example:
- "External perimeters (all)"
- "External perimeters (outside only)"

(could be reworded, and/or split differently - inside/outside, all walls/inside, etc. )
 
**Describe how it would work**
Each setting would affect different external perimeters, similar to Fuzzy skin which can be set to the outermost external perimeters or all external perimeters.

**Describe alternatives you've considered**
I tried mesh modifiers, but 
a) that's just not feasible (time and complexity-wise) if one slices tens of different shapes. 
b) it often won't just adjust the speed, it can also modify the existing perimeters producing undesired geometry/slicing changes

**Additional context**
Example print job that used nearly 1kg of filament.
Before:
[image]
After:
[image]

neophyl, 2023-03-01:
How do you determine which external perimeters are 'inside' or 'outside' ?    As a human you may look at an object and say that's obviously the inside but that's the sort of thing that requires AI to figure out (badly) unless its trained on a massive data set.

Misfi, 2023-03-01:
> How do you determine which external perimeters are 'inside' or 'outside' ? As a human you may look at an object and say that's obviously the inside but that's the sort of thing that requires AI to figure out (badly) unless its trained on a massive data set.

The functionality to differentiate inner/outer walls is already there:
[image]

closed 2026-09-07 by Misfi