prusa3d/PrusaSlicer #9850: Excessive extrusion rate

rcarmo, 2023-02-24:
### Description of the bug

I am using 2.6 alpha 4 and it is generating massively high extrusion rates when generating Marlin or RepRap G-Code.

Klipper stops with a safety guard like so:

```
Move exceeds maximum extrusion (28.274mm^2 vs 0.640mm^2)
```

Note that the extrusion rate it is generating is several orders of magnitude above the safe limit.

### Project file & How to reproduce

[double tunnel.zip](https://github.com/prusa3d/PrusaSlicer/files/10826876/double.tunnel.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha 4

### Operating system

Fedora 37

### Printer model

BQ Prusa i3 clone (MK2 equivalent)

rtyr, 2023-02-25:
You are using extreme value for bridge flow (default is 1, yours is 100). Using some sensible value will likely fix the issue.

jessegrunert, 2023-05-13:
I had this exact same problem, I figured it was a setting but couldn't figure out which one, thanks for this!