prusa3d/PrusaSlicer #9851: Color Change (manual) after 2nd layer not work, it change after first

Poke-Zwei, 2023-02-24:
### Description of the bug

Hi, I try to print thin object with just 6 layer for 1.2 mm high. I want first two layer in green, next two layer in white and last two layer in red. So I've not yet MMU2..so I do it manual. Prucer Slice produce g-code where the first color change/break is just after first layer, then the next occurs two layers later and then it print the rest of three layers in red. The printed model have 1.2mm high. So I removed 2.5.0 from my Mac (Ventura 13.2.1) also every file in general Library and user library , reinstalled 2.5.0 with same problem. I tried also 2.6.0 alpha 4 with same results. The g-code is little bit different between this versions, but first color change happens always around line#1820.

### Project file & How to reproduce

Project-File=  s1e-Fillet002.3mf 
STL-File = s1e-Fillet002.stl - done with FreeCAD 0.20.2

Reproduce: Just print it..color change break occurs after first layer..second color is printing in two layer and the last one is printed in 3 layers......or check out g-code when M600 happens around line#1830
[BUG - s1e-Fillet002.zip](https://github.com/prusa3d/PrusaSlicer/files/10828471/BUG.-.s1e-Fillet002.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 & 2.6.0 alpha4

### Operating system

Mac Ventura 13.2.1

### Printer model

Prusa MK3S+

Jan-Soustruznik, 2024-03-12:
Hello @Poke-Zwei, I think that you have added the color change at the wrong place. Command `M600 `is added to the start of current line. So you have to add change color at the 3th layer, then it will be print as you request. See attached screenshot. If you dont have any questions, please close this issue. Thank you, have a nice day 
[image]

Jan-Soustruznik, 2024-03-13:
Hello, Poke2, 
I agree that can be confusing, but if you give it more time. We have a knowledge base(https://help.prusa3d.com/article/general-info_1910) and also community forum(https://forum.prusa3d.com/) where is lot of information and advice of other users. 
Check the bar again there is show both for the last red color layers, one at 1.00mm(5) and second at 1.20(6).

I closed the issue as solved.

closed 2024-03-13 by Jan-Soustruznik