prusa3d/PrusaSlicer #9856: Slicer image disappears

Spatticus90, 2023-02-25:
### Description of the bug

After slicing several times the imaging for the file disappears I can't see the bed or anything I'm forced too restart the program 

### Project file & How to reproduce

[20mm_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/10830518/20mm_cube.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha r

### Operating system

window 10

### Printer model

ankermake m5

kubispe1, 2023-02-26:
Can you provide more info... 
- Descriptive images
- video, gif
I am not able to reproduce it on WIN10.
Thanks

Spatticus90, 2023-02-26:
When it happens again I been in the habit of closing it and reopening when
I'm done. It was from slicing changing something slicing again repeat
probably 5 times the computer is 6 years old so that may have something to
do with it

Spatticus90, 2023-02-27:
https://drive.google.com/file/d/1jCbLsiw6YyZlGf7bjEyHQiLzYJzCRqUG/view?usp=drivesdk
the video was too large so I put it in my drive hope you can see it

kubispe1, 2023-03-01:
Thanks for the additional video. That is useful. 
I see, there is also an issue with the screen that is cut.
- Does this happen also with 2.5.0?
- Can you additionally provide System Information (help/System Info)?
- and PC configuration?
Thanks for your time anyway...

Spatticus90, 2023-03-01:
I was not able to replicate it on 2.5 and what do you mean pc
configuration?

closed 2026-09-29 by prusaslicer-bot[bot]