prusa3d/PrusaSlicer #9858: Organic support

Rattlhead, 2023-02-25:
### Description of the bug

When generating, the support does not take into account other models.

### Project file & How to reproduce

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

MacOS

### Printer model

ulti steel

neophyl, 2023-02-25:
Please read https://github.com/prusa3d/PrusaSlicer/issues/9860 as I really cant be bothered to type it all out again.

kubispe1, 2023-02-26:
Closing this issue as a duplicate of others issues and there is a most original one: #316

closed 2023-02-26 by kubispe1

foreachthing, 2023-02-27:
@neophyl , @kubispe1 maybe fix (pin) an issue? 😉