prusa3d/PrusaSlicer #9862: PrusaSlicer 2.5.0 slicing issue

bux024, 2023-02-25:
### Description of the bug

please try to slice the attached file with any print setting preset.

on the plater it will be shown look like this:

[image]

after slicing, this is the outcome:

[image]

stl source: https://www.printables.com/model/274646-nutella-glas-candy-dispenser/files

### Project file & How to reproduce

[Handle-v2.zip](https://github.com/prusa3d/PrusaSlicer/files/10832075/Handle-v2.zip)

just slice.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha3 AND 2.5.0 stable

### Operating system

Win 11

### Printer model

MK3S+

JohnOCFII, 2023-02-25:
The model has errors.  See the triangle next to the model, and the note about 94 open edges.

[image]

I've run the STL through an online repair tool.  The repaired model is attached.  It has been compressed with a ZIP tool.  Unzip before using to get the repaired STL file.

[Handle-v2_fixed.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/10832179/Handle-v2_fixed.stl.zip)

bux024, 2023-02-25:
ooh thank you so much. i have closed the issue, sorry for the trouble.

closed 2023-02-25 by bux024