prusa3d/PrusaSlicer #9863: Organic supports - some support elements don't start from first layer.

QVirus, 2023-02-25:
### Description of the bug

Some organic support elements starts after 10 and more layers.
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Ukulele_Stand.zip](https://github.com/prusa3d/PrusaSlicer/files/10832113/Ukulele_Stand.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4+win64

### Operating system

Windows 10

### Printer model

Creality CR6-SE

neophyl, 2023-02-25:
Isn’t this just a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9601 ?  Which is still on this first page so it’s not even like it has to be searched for.

kubispe1, 2023-02-26:
I think it is mentioned duplicate issue. Thanks

closed 2023-02-26 by kubispe1