prusa3d/PrusaSlicer #9864: "Add Instance" with Multiple Objects Selected Causes Crash

Gunner087, 2023-02-25:
### Description of the bug

If multiple objects are selected at the same time and the "add instance" button is pressed, PrusaSlicer crashes.

### Project file & How to reproduce

Open project, select both objects, click "add instance", crash.

[Add Instance Crash.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10832198/Add.Instance.Crash.3mf.zip)

https://user-images.githubusercontent.com/65746045/221377455-6d27d4b7-a22b-41a2-862c-704409aa4487.mov

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

macOS 13.0.1

### Printer model

Voron0 v0.1

kubispe1, 2023-02-26:
Hi, thanks. WIN works fine to me PrusaSlicer-2.6.0-alpha4+19-win64.
Let's check MAC.

FidelCapo, 2023-02-27:
I am able to crash reproduce on macOS.

kubispe1, 2023-02-27:
OK, Win has the same issue. My mistake :D ....

FidelCapo, 2023-02-28:
Thank you for the report. This issue should be fixed in the next release.
Closing.

closed 2023-02-28 by FidelCapo