prusa3d/PrusaSlicer #9866: Placeholder total_layer_count wrong with supports

mjonuschat, 2023-02-25:
### Description of the bug

**What is the problem**

The placeholder [total_layer_count] is wrong when supports are being generated. Added a custom layer progress message to the before layer change gcode that exhibits the problem. In the attached project generating supports is enabled and the exported gcode shows messages  like `M117 Layer 1/228 : Prusament PLA`. The slicer UI shows 150 layers.

The correct number of layers is being exported without supports (150 in the UI and in the gcode files)

**What did you except**

Same layer number counts between UI and generated gcode.

Issue seem similar to issue #7028 but only happens with enabled supports

### Project file & How to reproduce

[Layer-Count-Bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10832293/Layer-Count-Bug.3mf.zip)
[image]

**How to reproduce**
1. Slice the attached project file
2. Note the number of layers in the UI
3. Export the GCode to disk
4. Open gcode file in editor, search for "M117 Layer 1"
5. Check the total layer count replacement (in my file it was 226 with supports enabled)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2+MacOS-arm64 / 2.6.0-alpha4+MacOS-arm64

### Operating system

MacOS Ventura 13.2.1

### Printer model

Prusa i3 MK3S

neophyl, 2023-02-25:
This is expected and correct operation. Supports have always generated with their own layer heights, they are not tied to the model at all.  Except for when using soluble supports. That is why the synchronise with layer height is greyed out normally. 

The placeholder is technically correct as there ARE that number of layers. Sounds like you want a new placeholder for total number of model layers.

mjonuschat, 2023-02-25:
Well there is technically correct and there is the issue of unexpected behavior based (minimal) documentation available and the output in the UI. 
I wouldn't know to request a new placeholder because I shouldn't need to know that supports generate their own layers nor does the UI give any sort of indication in regards to this behavior.
So for all intents and purposes this is a valid bug (that can be fixed in code or in documentation) and based on the outcome of that resolution there might be a feature request in the future to get a desired behavior.

mjonuschat, 2023-02-25:
Also lets be real: The gcode generater only generates 151 layer changes - the reasoning provided in your explanation is slighly flawed:
```
❯ grep -c 'M117 Layer ' Layer-Count-Bug_0.2mm_PLA_MK3S_1h27m.gcode
151
```

neophyl, 2023-02-25:
Except that the UI does indicate this.  The preview of a slice has the layer slider.  That displays the total layers.  It shows the different numbers depending on if support is enabled or not.  

The point about documentation may be valid, except there is this https://help.prusa3d.com/article/support-material_1698#synchronize-with-object-layers  

You might want to spend some time learning how the software actually works before calling it unexpected.  Its quite expected and has been that way since before it even was Prusa Slicer, as Slic3r does the same.

I'm not saying a different placeholder isn't warranted but the operation of the current one cant be changed without potentially breaking functionality for people who are actually using it in its current form.  

We seem to have different opinions on the difference between a 'bug' and a feature change request.

mjonuschat, 2023-02-25:
I have no desired to get into a slapfight with some random person on the internet. If you're a dev or a moderator that can declare this as wont-fix do so and I'll go away, otherwise I consider this a bug and have reported it and the powers to be can decide what to do with it.

And no, the UI does not make that differentiation as you can see in the attached screenshots - both of them show 150 layers in the slider.

[image]
[image]

neophyl, 2023-02-25:
Erm 2 images with the same settings of supports none ?  Please turn the supports ON and then slice and check your facts.

Just trying to pass on some info you seem to be unaware of.  But if you aren't interested in learning then have a good evening/day.

mjonuschat, 2023-02-25:
Please check the right side-bar with the object settings / support overrides and the inner display on the cube of the upper picture - it clearly shows supports inside the cube.

neophyl, 2023-02-25:
Then the layer count slider is quite possibly a bug when its with a modifier.  Try it without a modifier with the supports /on off in the profile.  The count changes at that point.  

I also added a line to my previous post with a response by a dev in a previous issue.

neophyl, 2023-02-25:
Congratulations, you have found a bug, the count isn't updated with the support layers when it is type organic it seems.  I just added a modifier to a test part I was looking at for someone on the Prusa forum and with type snug the count is different to the object layers and is showing the extra layers.  With Organic type the count is just the object layers.

bubnikv, 2023-03-03:
It is really not well defined what to put into total_layers variable,
because:
1) Each object could be printed with its own layering.
2) Supports may be printed at their own layer heights.

We may emit something like "total_object_layers", but that would not work
easily in case 1).

mjonuschat, 2023-03-04:
I'm actually less concerned about the actual value in there and more about the discrepancy in the output between layer_index and total_layer_num. With the current bug layer_index never reaches the total_layer_num value so even just increasing the layer_index every time a different "layer" gets started would be ok in my book.

Right now you end up with 150/228 being the final output. Either 150/150 for the final layer or 228/228 would be fine (for my purposes) - it doesn't necessarily have to be object layers.

bubnikv, 2023-03-06:
I see.
The problem is caused by the code that counts the layers, but it does not use an epsilon check for print Z of object vs. support layers.

Fixed for non-sequential prints with de3de3fd1177219c1177382abf7a7cd716d8969c
Fix for sequential print would require a lot more refactoring, I am postponing it for now.

mjonuschat, 2023-06-19:
This fix has been working well enough, closing the issue

closed 2023-06-19 by mjonuschat