prusa3d/PrusaSlicer #9868: Elegoo Neptune 3 Pro Thumbnail Support

nailuj29, 2023-02-26:
### Description of the bug

I just bought an Elegoo Neptune 3 Pro, and when using the profile in version 2.6a4, it does not show the thumbnail, instead only showing the Elegoo logo

[image]

### Project file & How to reproduce

[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/10832751/3DBenchy.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+linux-x64-GTK3

### Operating system

Arch Linux 6.1.12

### Printer model

Elegoo Neptune 3 Pro

bubnikv, 2023-03-03:
We leave it to Elegoo to provide a pull request implementing this feature.
Closing.

closed 2023-03-03 by bubnikv

fifonik, 2023-10-08:
There are number of scripts that fixing the thumbnail:
https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod
https://github.com/TheJMaster28/ElegooNeptuneThumbnailPrusa
https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa