prusa3d/PrusaSlicer #9869: Organic supports generate crossing perimeters

Matesaktesak, 2023-02-26:
### Description of the bug

When using the 'Arrange' tool to organize the platter and using the 'Organic supports' the slicer generates lots of crossing/overlapping extrusions/perimeters.

While this is not a huge problem, it rises the risk of failure and makes the supports more wasteful.

Shall further info be required, I'll try to respond ASAP, thanks!

### Project file & How to reproduce

[platter.zip](https://github.com/prusa3d/PrusaSlicer/files/10833460/platter.zip)
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Windows 10

### Printer model

Prusa i3 MK2

beelsebob, 2023-02-26:
Yup, seeing the same issue here too.  Attached another 3mf with far fewer crossing perimeters just in case it's easier to debug.

[Spool Winder Parts.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10833675/Spool.Winder.Parts.3mf.zip)
[image]

foreachthing, 2023-02-26:
Starting here: https://github.com/prusa3d/PrusaSlicer/issues/9835
and then quoting [neophyl](https://github.com/neophyl):
> All supports and even other objects have always done this in PS. Its expected operation. It just more noticeable now that organic supports are located further from the object they are supporting. Each object is sliced individually. Place 2 over lapping objects and you will have a very bad day. You as the user are expected to avoid this.
> 
> To avoid this the correct procedure is to merge the objects. In this way PS treats them as one and then supports etc do not ignore them. This has been reported many many times already, they are closed as they are duplicates.
> 
> For example the most recent closed one #9723 or #9514 #8273 #7697

beelsebob, 2023-02-26:
Just because it's the behaviour today, doesn't mean it's the "correct" behaviour.  Expecting the user to magically know where the slicer is going to think to put the supports is a terrible user experience.  In my example, there were plenty of places the slicer could have chosen to place the support not interfering with the neighbouring object, but it just happened to place them in such a way that it landed in the 180° region it shouldn't, not the 180° region it could.

While I get the "each object is sliced individually" stance - that sounds like it simplifies things a lot, it seems like there's a reasonable compromise that could be made in giving the slicer a polygon in which it's allowed to place supports, or better yet a volume.

I also understand that that kind of approach would lead to the result of slicing depending on the order the objects in the scene are sliced, but it seems like it's a reasonable compromise given that the alternative is to expect the user to be psychic.

beelsebob, 2023-02-26:
A further thought on this.  Perhaps another option would be to allow the user to specify support blockers on the bed, as well as on the model.  If I were able to tell the slicer "don't place supports on top of this other part", it would allow me to fix issues like this one, and it would mean that there was no longer a dependency on order of slicing.

Matesaktesak, 2023-02-26:
I can take the arguments made about user having the objects far enough apart. However, the example I have given was generated by tho autě Arrange tool. One or the other has to be fixed.

foreachthing, 2023-02-26:
> ... However, the example I have given was generated by tho autě Arrange tool. One or the other has to be fixed.

Auto-arrange can be adjusted, meaning more or less offset. Adjust it to, maybe 10 mm (or even more), and try again.

kubispe1, 2023-02-26:
There are a few collision types that persist in PS from the beginning of his existence. It is not so easy to eliminate and prevent them, even if it looks trivial from a user's point of view. Slicing of individual objects that cause this issue brings other advantages. In the long term run, we will need to improve it. 
Auto-arrange is a separate function and has no idea about support generation at the moment. :-( 
Thanks for understanding
Closing as duplicate: #316

closed 2023-02-26 by kubispe1