prusa3d/PrusaSlicer #9870: Alpha 3 and 4 freezes macbook pro 13" 2022 with intel i5

pejspa, 2023-02-26:
### Description of the bug

- open prusa slicer alpha 3 or 4. 
- add any .stl file (size does not matter)
- frezzing and throttling whole time the program is running. 

### Project file & How to reproduce

- open prusa slicer alpha 3 or 4. 
- add any .stl file (size does not matter)
- frezzing and throttling whole time the program is running. 

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4+MacOS-x64

### Operating system

MacOs Ventura 13.2.1 (22D68)

### Printer model

prusa mini+

FidelCapo, 2023-02-27:
Is this behavior different between 2.5.0 and 2.6.0 alpha versions?

nigeljohnson73, 2023-03-02:
I have an Intel iMac Pro

[image]

It appears to hang once you generate a complicated model that fills the print volume. (using organic supports) and using the "center of gravity" indicator that I've never used before. Switching off the CoG does make it 'faster' but it still will not drag properly - a couple of seconds between refreshes. There is no CPU thrashing.

Organic supports did not exist in 2.5 so for me this is different. Regular supports seemed ok, and this got worse when I chopped the model in 2 so my head (and beard complexity) was bigger.

[image]

Here is the model file:

[Nigel Bust slow.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10868270/Nigel.Bust.slow.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]