prusa3d/PrusaSlicer #9871: alpha 4 crashed on Mac OS

lnjustin, 2023-02-26:
### Description of the bug

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [704]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021 (PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-02-26 09:31:27.1115 -0500
OS Version:            macOS 12.6.2 (21G320)
Report Version:        12
Anonymous UUID:        4B9E5EB3-90BE-1363-CC6F-F
[McGonagall Full copy.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10833926/McGonagall.Full.copy.3mf.zip)
E6744253A15

Sleep/Wake UUID:       25A917D4-93EC-400B-9B66-90B8D8B475E8

Time Awake Since Boot: 110000 seconds
Time Since Wake:       68531 seconds

System Integrity Protection: enabled

Crashed Thread:        6  slic3r_BgSlcPcs

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 8 Floating point exception: 8
Terminating Process:   exc handler [704]

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80fccc97a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80fcccce8 mach_msg + 56
2   CoreFoundation                	    0x7ff80fdd036d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff80fdce9f8 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff80fdcde3c CFRunLoopRunSpecific + 562
5   HIToolbox                     	    0x7ff818a7d5e6 RunCurrentEventLoopInMode + 292
6   HIToolbox                     	    0x7ff818a7d34a ReceiveNextEventCommon + 594
7   HIToolbox                     	    0x7ff818a7d0e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
8   AppKit                        	    0x7ff812807fad _DPSNextEvent + 927
9   AppKit                        	    0x7ff81280666a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
10  AppKit                        	    0x7ff8127f8d19 -[NSApplication run] + 586
11  PrusaSlicer                   	       0x1113a5ffa wxGUIEventLoop::OSXDoRun() + 186
12  PrusaSlicer                   	       0x11149129d wxCFEventLoop::DoRun() + 29
13  PrusaSlicer                   	       0x111407cd5 wxEventLoopBase::Run() + 85
14  PrusaSlicer                   	       0x1113dfa24 wxAppConsoleBase::MainLoop() + 116
15  PrusaSlicer                   	       0x111365d4a wxApp::OnRun() + 26
16  PrusaSlicer                   	       0x111439379 wxEntry(int&, wchar_t**) + 73
17  PrusaSlicer                   	       0x10f450938 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 248
18  PrusaSlicer                   	       0x10e85acb7 Slic3r::CLI::run(int, char**) + 21159
19  PrusaSlicer                   	       0x10e85fc30 main + 352
20  dyld                          	       0x11de2452e start + 462

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	    0x7ff80fccc97a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80fcccce8 mach_msg + 56
2   CoreFoundation                	    0x7ff80fdd036d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff80fdce9f8 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff80fdcde3c CFRunLoopRunSpecific + 562
5   AppKit                        	    0x7ff8129759ce _NSEventThread + 132
6   libsystem_pthread.dylib       	    0x7ff80fd094e1 _pthread_start + 125
7   libsystem_pthread.dylib       	    0x7ff80fd04f6b thread_start + 15

[... 326 lines cut ...]
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB30Bus
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

### Project file & How to reproduce

[McGonagall Full copy.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10833927/McGonagall.Full.copy.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha4

### Operating system

macOS 12.6.2

### Printer model

Prusa Mini+

FidelCapo, 2023-02-27:
Unfortunately I am not able to reproduce the crash on macOS. Can you please attach a video or all the steps that need to be done to make PrusaSlicer crash? Thank you.

bubnikv, 2023-03-03:
Crashes in  Slic3r::FFFTreeSupport::generate_support_areas()
We will be thankful for a project file to reproduce.

kubispe1, 2023-03-23:
@lnjustin , can you provide 3mf please?

lnjustin, 2023-03-23:
> @lnjustin , can you provide 3mf please?

I think it’s already in the post
https://github.com/prusa3d/PrusaSlicer/files/10833926/McGonagall.Full.copy.3mf.zip

kubispe1, 2023-03-23:
Ok, sry I see it. 
I am also not able to reproduce it. I test it on Mac for 
2.6.0-alfa4
2.6.0-alfa5
2.6.0-alfa5+181

closed 2026-09-29 by prusaslicer-bot[bot]