prusa3d/PrusaSlicer #9876: Organic Supports don't work with Height Range Modifier

dschwick, 2023-02-26:
### Description of the bug

When using a height range modifier to change layer height, organic supports go through the unit.

I'm using a 0.3mm layer height, then modifying the layer height of the curved part of the model to be 0.1mm. It looks like the slicer calculates that there will be more layers (99 -> 400) but the supports get confused and don't know how to adjust. Snug and Grid supports work correctly in this scenario.
[image]

### Project file & How to reproduce

1) Open .stl with prusaslicer 2.6.0 alpha (I'm using an ender 3 bed handle, thing:2957507)
2) Set layer height to 0.30mm
3) Set supports to "on build plate only"
4) Print Settings -> Support Material -> Section: "Options for support material and raft" -> Dropdown: "Style" -> Set to "Organic"
5) Plater -> Right side menu -> Right click handle_ender3.stl -> Height range modifier -> Mine is set to: Start at height=4.1, Stop at height=30, Layer height=0.1mm)
6) Slice
7) Support goes through model
[ender cooling ducts.zip](https://github.com/prusa3d/PrusaSlicer/files/10834845/ender.cooling.ducts.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha2

### Operating system

Windows 10

### Printer model

Prusa MK3S+

neophyl, 2023-02-27:
Please read the release notes for the alpha software.  They specifically mention that organic supports do not support the use of variable layer heights.  They currently have several limitations, like rafts too for example.

dschwick, 2023-02-27:
Ok, thank you. Just wanted to make sure you guys knew!

kubispe1, 2023-03-01:
In the actual 2.6.0 release Variable layer height and height range modifier will be disabled (with high probability). But we see that many users use it together.... due to comments as mentioned @neophyl :D :D

dschwick, 2023-03-01:
> Please read the release notes for the alpha software. They specifically mention that organic supports do not support the use of variable layer heights. They currently have several limitations, like rafts too for example.
> 
> It has been reported multiple mulitple times. So much so that alpha 3 doesn't let you do that combination there were that many reports from people who did not read the release notes.

I see you edited your comment with that second paragraph after my initial reply. It seems like I really hit a sore spot for you here. I hope I didn't ruin your day too badly. I'm new to github and tried searching for similar bug reports but found nothing. I'll be sure to be more thorough with my search in the future. Thanks again!!

dschwick, 2023-03-01:
> In the actual 2.6.0 release Variable layer height and height range modifier will be disabled (with high probability). But we see that many users use it together.... due to comments as mentioned @neophyl :D :D

Great, thanks for the heads up! I'm very happy with organic supports though so hopefully a fix gets figured out.

closed 2023-03-01 by dschwick

kubispe1, 2023-03-02:
There is the original thread for the issue: #9462