prusa3d/PrusaSlicer #9881: Failure to Eject Drive after exporting G Code.

Bladevane, 2023-02-27:
### Description of the bug

After exporting the G Code to an SD Card the drive fails to eject when the "Eject Drive CTRL T" icon is selected. See screenshot for failure message.

Error message appears "Ejecting of device PRISA(H:\) has failed".

### Project file & How to reproduce

[Filament Feed Adaptor.zip](https://github.com/prusa3d/PrusaSlicer/files/10841498/Filament.Feed.Adaptor.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 4.

### Operating system

Windows 11

### Printer model

Prusa Mk3S and Caribou Mk3S

kocikdav, 2023-03-02:
Hello. Thank you for reporting this issue.  We have changed our code for ejecting on Windows for 2.6.0. Unfortunately we are still having issues so I would like to ask you some additional information.
1. Is your SD card drive external or internal? External is usually connected via usb while internal is directly on chassis of PC. (Some box case has actually external drives though)
2. Would you run PrusaSlicer 2.6.0 alpha-4 from prusa-slicer-console.exe? There should be a error message with some additional information.

Bladevane, 2023-03-02:
Thank you for your response.
1. SD Card is connected using an external drive via USB. I have also tried the SD card in the PC slot and the error message is the same.
2. I initially ran Prusa Slicer from the file prusa-slicer-console.exe but now I have the icon on the taskbar and run it from there. I tried both methods today and there was no error message from either method. I did notice this when running from the console.exe file [2023-03-02 10:25:14.329811] [0x0000380c] [trace]   Initializing StaticPrintConfigs

Regards,
Glyn.

MisnerJ, 2023-03-04:
Adding more information to this report. I noticed external card readers become completely disconnected using "Eject drive Ctrl+T" in 2.6.0-alpha4+win64. Might be related to a failed ejection in this case?

On my machine I am able to achieve "Successfully unmounted." with the Filament Feed Adapter project file here, but the external card reader completely disconnects. Successive SD cards are not able to be mounted without unplugging/replugging the reader. Tried it with a USB-C reader and a USB2.0 reader.

If this is unrelated I can start a new issue.

Bladevane, 2023-03-04:
I tried ejecting from the external drive (failed every time) and the
in-built drive on the PC (also failed) so I am assuming it is a glitch in
the coding with version 2.6.0 as 2.5.0 works fine.

bubnikv, 2023-03-06:
The problem there is that a USB device may or may not identify itself as a device that supports ejecting a medium. Then it is not clear whether the user wants to detach the whole USB device or just to eject a medium from the USB device. We used to just eject, but that does not satisfy some users. Now we are disconnecting the USB device.

Unfortunately the devices do not seem to provide correct information to operating system as of whether the application should force media eject or disconnect of the whole USB device. Even Windows do it wrong on some USB SD card readers, they eject the whole reader.

We don't know how to improve the situation. We may either roll back to 2.5 behavior or keep the 2.6 behavior, neither will make  everybody happy. We would be thankful for a guidance on how we can decide in which case to eject and in which to disconnect.

Bladevane, 2023-03-07:
Thank you for the full explanation. Personally I see the issue as detrimental and would like to see a return to 2.5.0 functionality but recognise others may feel differently. The lack of the facility is not crucial and I have already changed my way of working to ejecting the drive using Windows. Regards.

vistax64, 2023-03-07:
it may sound silly, but what about having 2 eject buttons, even if you call them eject 1 and eject 2, users who are impacted negatively from 2.5 -> 2.6 eject function would pretty easily know which one suits their needs, and conversely anyone who was impacted in 2.5 but sees improvement in 2.6 eject functionality will chose the eject button that works for them.  that way users would just know which eject works for their needs, and if in the future say eject 1 fails, they can simply use eject 2 going forward.  its not a great user experience for sure, having 2 eject buttons, but that seems like it at least solves all issues and can be a band aid fix until a more elegant approach can be found.

[image]

Bladevane, 2023-03-07:
I like the idea. Can this be implemented by the Prusa Slicer team?

robot256, 2023-03-25:
I am also having this issue with 2.6.0alpha5 on Windows 11.  I am using a Sabrent USB 3.0 SD/microSD card reader.  Mostly it fails, sometimes it succeeds (and I don't remember, but I probably had to replug it afterwards).  On 2.5 it never failed, and only occasionally would I need to replug the USB adapter in order to make a reinserted SD card appear.

I agree with others that the 2.5 behavior was better and I would hate for everyone with correctly-functioning devices to have to replug their USB ports all the time.  Perhaps an option in the settings for "USB reader eject compatibility mode" would users with misbehaving devices?

kocikdav, 2023-03-31:
Hello recently published alpha6 has commit 6be2aa29dc830bddf22f30a54dcbc12e8a3613f1, that uses the old eject code for external card readers or if the new eject function fails. We cannot only rewind to the old function since it had issue of ejecting too soon. But we are looking to fix it too. Please try that your external reader is not ejected whole any more. If the troubles still remains, please run prusa-slicer-console.exe and paste here the log. Since there are many devices on the market, it is possible some behave different than we expect.

Bladevane, 2023-04-03:
Ejecting still failed. Here is the copy of the log.

 

[2023-04-03 10:09:59.607317] [0x000062ec] [trace]   Initializing
StaticPrintConfigs

[2023-04-03 10:10:32.735342] [0x000062ec] [error]   Ejecting of H:\ has
failed: Request to eject device has failed. Veto type: 8

[2023-04-03 10:10:32.735342] [0x000062ec] [error]   Ejecting has failed.

kocikdav, 2023-04-03:
@Bladevane My apologies, in your case the log that would help is on warning level. Can I ask you to run prusa-slicer-console.exe with --loglevel=2 parameter? To do so, you need to open Terminal (right click File explorer inside alpha6 folder -> Open in Terminal ) and type `.\prusa-slicer-console.exe --loglevel=2` . Now after eject there should be another line _Ejecting of H:\ has failed: Request to eject device has failed. Veto type: X_. Please let us know what the veto type is.

Bladevane, 2023-04-04:
Thanks for the advice. This is the message I get.

Veto type: 8

 

Regards,

 

Glyn.

jfinnie, 2023-04-11:
I created that issue https://github.com/prusa3d/PrusaSlicer/issues/10283 as I've been seeing issues with my internal card reader becoming unusable after ejecting until a reboot. 

I've just done some experiments with an external card reader and I think the behaviour there is almost as broken.

It seems to me to be wrong to be ejecting the whole USB device that parents the disk.  An example which shows how wrong this is; I have an external multi-slot card reader make by Kingston, part number FCR-HS219/1.  It has several slots that each receive their own drive letter.  I did a test and put drives in several slots.  If I export gcode to a drive in one of the slots and then eject; Prusaslicer now ejects the whole USB device, unmounting all of the SD cards that were in use in that reader; the Prusaslicer UI then tells you that only the device it was concerned with was unmounted.  

This has to be very undesireable.  Operations should be limited to the volume in question and not to the USB device.  I didn't ask it to eject the entire USB device including any other mounted drives.

Prusaslicer 2.5 behaves as you'd expect with such multiple slot SD card interfaces - only the SD card in question is ejected, the others are left mounted.

kocikdav, 2023-04-17:
Hello everyone, please try this build:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha6+85-win64-gfb6dd330c-202304170852.zip
We have switched to completely different code for ejecting. During our testing, it seems to work correctly and even triggers Windows notification so we hope, all goes well this time.
Please note, this build is experimental and for purposes of testing the eject function. Other features might be broken or work in progress. Thanks.

Bladevane, 2023-04-17:
Thank you for your prompt action. I downloaded the folder and it seems to
work very well.

 

Thank you,

 

Glyn.

jfinnie, 2023-04-17:
This works well for me.

MisnerJ, 2023-04-18:
Also working well for me! Tested on two machines with multiple readers. No issues.

Machine Builds: Windows 10 Home 19044.2846, Windows 11 Pro 22000.1574
External Readers: SanDisk ImageMate PRO SDDR-A631, SONY Multi-Card MRW62E, BOWER All-in-1
Internal Reader: Surface Book 2

kocikdav, 2023-04-20:
Thank you everybody.

closed 2023-04-20 by kocikdav