prusa3d/PrusaSlicer #9884: Separated object brim

lucaspsara, 2023-02-28:
### Description of the bug

So when you turn on the **complete individual objects** function, while also having **Brim** in prusaslicer 2.6 on the first layer makes all the brims and then proceeds to finish the first part. 
I think it would be better if just made the brim when the is part is started, or it would cool to see the option to turn on this method.

Also nice to see the option to select to which part use brim and which do.

Thanks for reading, lucas

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

windows 10

### Printer model

e3v2 biqu h2 klipper

Snuff1eupagus, 2023-02-28:
That's nothing. You should look what happens if you try and add a skirt to those individually printed objects.

dajomas, 2023-06-02:
Using PrusaSlicer-2.6.0-beta3+win64-202305261352 and I have the same issue. When brims are printed for all objects first you'll loose the functionality of the brim (prevent warping) as it will start to warp before the object it is supposed to hold, is printed.

This is a bug that really needs fixing.