prusa3d/PrusaSlicer #9885: Monotonic lines allow z hop which may increase the infill duration

MrBlackmidi, 2023-02-28:
### Description of the bug

A little while ago, i discovered that the monotonic lines top or bottom infill allows z hop, especially if it has a large jump between the lines

### Project file & How to reproduce

[image]
[HexScraperPocket.zip](https://github.com/prusa3d/PrusaSlicer/files/10845528/HexScraperPocket.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 4

### Operating system

Windows 11

### Printer model

Prusa MK3

MrBlackmidi, 2023-03-12:
Hi, are you guys aware of this? Because I haven't received any updates on this.

MrBlackmidi, 2023-03-20:
@lukasmatena ?

MrBlackmidi, 2023-04-02:
Hey guys, are you all aware of #10066 #9672 #9285? because i havent recieved any updates on them.

MrBlackmidi, 2023-04-03:
guys?

amatulic, 2023-04-24:
I don't usually see comments from anyone who has the ability to prioritize issues, review PRs, and merge fixes submitted by the community. There are over 100 issues open, some of them several years old. I submitted a PR to fix an old issue #1207 more than two months ago and it seems nobody has looked at it.

This issue about z-hop with monotonic lines is unlikely to be noticed by an average user and isn't causing anyone actual damage (ruining prints), so I expect the priority would be low. What goes into version 2.6 has likely already been determined, so your issue and mine are probably going to wait for a future release.

closed 2026-09-29 by prusaslicer-bot[bot]