prusa3d/PrusaSlicer #9886: 2.6.0-alpha4 Elegoo Neptune 3 MAX gcode failure

bobditts, 2023-02-28:
### Description of the bug

gcode generated causes failures no matter which model is used. The gcode attached in the zip file fails during the last few layers (I would estimate less than 10 remaining layers). The failure that occurs is the print head stops printing mid print on the layer (nozzle stops and rests on the print). The machine reboots itself and returns to the main screen awaiting instruction. 

### Project file & How to reproduce

[32mm_NightmareDiceTower_4_18h0m.zip](https://github.com/prusa3d/PrusaSlicer/files/10845967/32mm_NightmareDiceTower_4_18h0m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Win10 v10.0.19045 build 19045

### Printer model

Elegoo Neptune 3 Max

rtyr, 2023-02-28:
I am afraid this will be more suited for Elegoo tech support.

FYI @mlee12382

mlee12382, 2023-02-28:
@rtyr agreed, that does seem to be more of a printer issue than slicer issue. If it's happening at a specific location each time maybe hardware related ie loose connection or intermittent cable termination or the like. Could also be something firmware specific. My advice would be to attempt to isolate the specific area of failure and see if it repeatable across multiple prints and try a different sd card since the included cards can sometimes have issues. Also try printing normally but connected to pronterface and see if it spits out an error  that might help nail down the problem.

bobditts, 2023-03-01:
My apologies. I left out a lot of information. I have tried multiple memory cards, slicers, models. Only time the prints fail (doesnt matter which model) is when sliced with prusaslicer. The fail will replicate itself with the same gcode in the exact same location, but will not duplicate identically on different models.

mlee12382, 2023-03-01:
@bobditts is it failing on specific types of features? Maybe there's a specific gcode command being issued that the max does recognize? I'd still be curious to know what if any feedback the printer gives through a terminal at the time of failure. Are there any error codes on screen?
Also something quick to check, make sure the gcode flavor is set to either Marlin 2 or Marlin legacy in the slicer printer settings.

rtyr, 2023-03-01:
I think you should still contact Elegoo tech support and provide them some of the problematic g-codes. If printer cannot handle standard g-code, than it should be likely fixed on the firmware side.

neophyl, 2023-03-01:
I'd be interested in this being fixed as the Max was the next printer I was looking to buy when they become available again.  

Commenting so I can track this :)

closed 2026-09-29 by prusaslicer-bot[bot]