prusa3d/PrusaSlicer #9888: Reload from disk on modifiers doesn't work

DeflateAwning, 2023-02-28:
### Description of the bug

Reload from disk on modifiers doesn't work to reload the STL content. Follow repro.

### Project file & How to reproduce

1. Import an STL.
2. Right click on STL.
3. "Add modifier" > "Load ..."
4. Select another STL
5. Right click on the modifier item in the far-right pane
6. Click "Reload from disk"
7. Observe that the preview did not update to the latest model, indicating that it wasn't reloaded. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Prusa (N/A, haven't even tried printing)

kubispe1, 2023-02-28:
I made quite a lot of exports and modifications with Fusion to test Reload from disk and it works. Even if you close Slicer and reopen Reload from disk works (That made some issues in past)...
here is short example .... 2.6.0 alfa4 and 2.5.0 works same.
- in case provide some STLs  - in the same way as in the gif
- or specify what did you differently...
[image]

DeflateAwning, 2023-02-28:
I was unable to reproduce this again just now, and I tried a few times. I think it's possible that I was saving to a different file or something Sorry to waste your time!

What'd you use to create that screen recorded gif?

kubispe1, 2023-03-01:
I use some LICEcap app. 
So, I am closing this issue. In case you will discover the issue. Let us know.

closed 2023-03-01 by kubispe1