prusa3d/PrusaSlicer #9890: Organic Support Tree Suspended Loops

RickR74, 2023-02-28:
### Description of the bug

When the tree support was building up for this model, it created two loops that were in mid air at a certain point, under an overhang. As seen by the screenshot.
[image]

### Project file & How to reproduce

[Tree Support Failure.zip](https://github.com/prusa3d/PrusaSlicer/files/10849828/Tree.Support.Failure.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

Edition	Windows 11 Home Version	22H2 Installed on	‎30/‎09/‎2022 OS build	22621.1265 Experience	Windows Feature Experience Pack 1000.22638.1000.0

### Printer model

Mini+

RickR74, 2023-02-28:
Just an update showing a better view of the un-connected loops!
[image]

theepicflyer, 2023-03-02:
I am seeing the same issue!

Supports are being drawn in mid-air underneath some overhangs.
Screenshot:
[image]

Project File:
[Cockosaurus_V3.zip](https://github.com/prusa3d/PrusaSlicer/files/10867230/Cockosaurus_V3.zip)

### Version of PrusaSlicer
Version 2.6.0-alpha4+win64

### Operating system
Windows 11

RickR74, 2023-03-02:
Well I thought the problem was related to the tiny size of my model, which is only 5mm wide and 10mm high.
But I see you (Theepicflyer) have the issue with a larger model.
I even loaded my model into PrusaSlicer version 2.5, which doesnt of course have organic supports, but I activated Snug supports and still had the same problem, as seen in the image attached.
That support was suspended 0.3mm above the surface below it with a layer height of 0.1mm and a bottom support clearance of 0.15mm
[image]

kubispe1, 2023-03-02:
In support material settings for Organic supports are these parameters - Maximum Branch Angle and Prefered Branch Angle. That helps me to solve the problem. AFAIK it is the only solution for this specific example.

kubispe1, 2023-03-02:
There it is:
[image]

kubispe1, 2023-03-02:
I find quite a similar issue #9838, where those angles are not appropriate.

RickR74, 2023-03-02:
Thanks for that info Kubispe1. OK so its not a bug, but settings. Should have played with those angles, but as new to organic supports I didnt realise the angle could cause that issue. I tried increasing the angle, but had to go up to 60 degrees before the branch grew across in one piece. So glad that was such an easy fix, cause I love these tree supports as I always found it a problem when trying to remove the bottom of supports from the model inside holes.
[image]

RickR74, 2023-03-02:
> From "kubispe1" ***@***.***>
> To "prusa3d/PrusaSlicer" ***@***.***>
> Cc "RickR74" ***@***.***>; "Author"
> Date 2/03/2023 11:13:17 PM
> Subject Re: [prusa3d/PrusaSlicer] Organic Support Tree Suspended Loops
> (Issue #9890)

Thanks for that information. It definitely sorted the problem out. I 
just wasnt aware the angle was so critical, but will keep an eye on that 
setting from now on.😀

closed 2023-03-02 by RickR74

KilianB, 2023-03-03:
While this might be fixable via altering settings I still feel this behaviour isn't really friendly to the user. 
It would be helpful to display a warning message which explains that generated supports are not fully printable and they might need to tweak the settings to alleviate the situation.

RickR74, 2023-03-03:
I Agree!

From "Kilian Brachtendorf" ***@***.***>

Cc "RickR74" ***@***.***>; "State change" 
Date 3/03/2023 9:03:23 PM

kubispe1, 2023-03-08:
It is not an excuse just a suggestion for now. We will try to keep it in mind but, honestly there we have more priority issues. Thanks for understanding.