prusa3d/PrusaSlicer #9891: Arrange overlaps organic support

fornellas, 2023-02-28:
### Description of the bug

When we have multiple parts, and ask to arrange them (A shortcut), it puts them too close together, and the generated organic supports overlap:

[image]

### Project file & How to reproduce

[organic support overlap.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/10849844/organic.support.overlap.3mf.gz)

- Open project
- Press "A" (Arrange)
- Go to Preview and spot the overlapping organic supports.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha4+linux-x64-GTK3-202302180033.AppImage

### Operating system

Ubuntu 22.04.2 LTS

### Printer model

MK2.5S

dan-and, 2023-03-01:
Also when support are overlapping, they move below the other model. 

Can be seen in fornella's settings: 
[image]

But even worse, when the expansion is larger, like 5mm (instead of fornella's 3mm expansion) 

[image]

kubispe1, 2023-03-02:
Instances behave similarly to individual objects. So collisions are there same.
Closing as duplicate: #316

closed 2023-03-02 by kubispe1