prusa3d/PrusaSlicer #9892: bad function call without any obvious reason

Agent-Cain, 2023-02-28:
### Description of the bug

I noticed some weird behaviour without any obvious reason. 
I exportet two different sized oreo models from my CAD using the same function. 
I made the same steps on both: First input, then automatic support painting and then slicing. 
The bigger one works without any problem or something, the smaller one won't slice. 
Added both projects to the zip file ... 
[image]

### Project file & How to reproduce

[error.zip](https://github.com/prusa3d/PrusaSlicer/files/10849951/error.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0a4

### Operating system

Windows 10

### Printer model

Voron V0

kubispe1, 2023-03-01:
I can confirm the issue in PrusaSlicer-2.6.0-alpha4+67. 
For "errors.3mf", I tried to repair the model and apply automatic support painting again. The issue is still there.

nexusgx, 2023-03-17:
I can confirm as well. I reproduced the issue with a different model and trying to print to a Lulzbot Workhorse.

[5mm adapter T.zip](https://github.com/prusa3d/PrusaSlicer/files/11001916/5mm.adapter.T.zip)

kubispe1, 2023-03-20:
Hi, we are looking into it. Thanks
SPE-1605

ADXGlock, 2023-04-13:
I have had the same issue when trying to slice these two models  (Currectly using 2.6.0 alpha6).
https://www.printables.com/model/132447-cable-soldering-jig
https://thangs.com/designer/models/3d-model/802663

Both have similar geometry.

FidelCapo, 2023-05-02:
This issue should be fixed and fix will be part of next release.
Closing.

closed 2023-05-02 by FidelCapo