prusa3d/PrusaSlicer #9893: Dual Extrution - can't print Skirt with other than first extruder

gbkwiatt, 2023-02-28:
### Description of the bug

Using IDEX printer with 2 extruders. I have a model with dual color using 2 extrodures.
In `Print Settings -> Multiple Extruders -> Extruders -> Support material/raft/skirt extruder: ` Doesn't matter if I select 1 or 2  it prints skirt with  first extruders. But I actually want to start the print with a skirt using second extruder. 

### Project file & How to reproduce

[211124_tensegrity-hexagons_2.zip](https://github.com/prusa3d/PrusaSlicer/files/10854456/211124_tensegrity-hexagons_2.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-aplha4

### Operating system

Windows 11

### Printer model

Sovol SV04

Bully85, 2023-05-03:
i have the same issue

d9C89, 2023-05-03:
me two

closed 2026-09-29 by prusaslicer-bot[bot]