prusa3d/PrusaSlicer #9895: First layer above aligned rectilinear infill is in the same direction as the infill, resulting in longer bridges

timur-tabi, 2023-03-01:
### Description of the bug

The first layer (of the top layers) that prints above aligned rectilinear infill prints in the same direction as the infill, which requires longer bridging.  This significantly reduces the benefit of the infill.

When using aligned rectilinear infill, the infill always fills in one direction, no matter the layer.  The actual angle is specified by the "Fill angle" option on the Advanced pane of the Print Settings page.  So every layer of the infill looks like this:
[image]

The problem is that the first of the top layers prints in the same direction:
[image]

This means that almost every line of that first layer needs to span the entire width of the plane without any support from the infill.

Changing the "Fill angle" to any other value doesn't help, because the first layer will always print at that same angle!

Instead, the first layer should print perpendicular to the "Fill angle".

### Project file & How to reproduce

[Mini_Jambos_Holder_V1.zip](https://github.com/prusa3d/PrusaSlicer/files/10856438/Mini_Jambos_Holder_V1.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+linux-x64-GTK3

### Operating system

Ubuntu 22

### Printer model

MK3s

timur-tabi, 2023-03-01:
I just noticed this text on https://help.prusa3d.com/article/infill-patterns_177130#infill-types-and-their-properties under "Aligned Rectilinear"

> However, using this infill might cause some trouble when the direction of the lines in the infill is the same as in the infill of the first top solid layer – if they are perfectly parallel, the top layers might have issues with bridging.

The problem is that the first top solid layer is _always_ in the same direction.  There's nothing you can do to fix it.

timur-tabi, 2023-03-01:
Also note that regular Rectilinear infill does not have this problem.

neophyl, 2023-03-01:
I thought that was the whole purpose for aligned rectilinear.  I think it was designed for transparent prints where you want **everything** printed in the same direction.
I suspect its designed to work in exactly the fashion.

timur-tabi, 2023-03-01:
Hmmm.   I mean, I guess that's what the "aligned" in Aligned Rectilinear stands for, but the layer right above it is printed perpendicular.  So not everything is aligned.

Not only that, but the infill is not aligned with the bottom layers.  The topmost bottom layer is printed perpendicular to the first infill layer.

So I'm not really buying that argument.  The help page for infill types makes no mention of transparent prints.

neophyl, 2023-03-01:
It’s supposed to be used with zero top and bottom layers. And at high infill percentages. It was added for a rather specific purpose. As mentioned above for getting the most chance at a translucent print.  If you want normal then don’t use aligned rectilinear. It’s also a relatively recent addition and PRUSA do tend to lag behind in documentation so the fact there’s no mention of that is not a factor to be relied on.

timur-tabi, 2023-03-07:
It's fixed, but only if the fill angle is 45°.  In that case, the first layer is printed at 135°.

However, if I change the fill angle to 135°, the first layer is still printed at 135°.  I suspect that this is a bug in the new bridging algorithm.

Godrak, 2023-03-07:
Yes, you are right. If the bridging infill fills the whole surface, it does not take into account the sparse infill. We'll look into it for the next alpha.

lukasmatena, 2023-04-02:
Fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena