prusa3d/PrusaSlicer #9898: Option to disable solid infill.

paulluby, 2023-03-01:
Hi Guys.

Is it finally time to include an option to disable Solid Infill?

I print Pilots for my Radio Control Model Aircraft and as I'm sure you can imagine, weight is everything.

The inclusion of Organic supports has made printing these pilots much easier, especially regards post print preparation.

However Solid Infill still remains the most frustrating thing when printing anything.

For me it adds time and more importantly weight to any Pilots I print.

Lets not mention the print failures it can cause by putting extra heat into an already hot area.

I print my Pilots using 5 upper and 5 lower layers, 3 perimeters and Lightening infill at 15%. This gives a good quality print but the Solid Infill can can add a lot of weight on a larger scale Pilot.

The pictures below From PRUSA Slicer are of a 1/6 scale pilot and as you can see Solid Infill increases print time by 16% and printed Pilot weight by approximately the same amount. The picture of the painted Pilot is a 1/3 scale pilot for comparison.

I use modifiers to prevent the solid infill in some areas but I require lightening infill in some areas to make the print possible, which also introduces Solid Infill in places where its not needed nor wanted as well as increasing the possibility of allowing it to generate infill that will be printed in mid air if the modifier is not placed perfectly.

I've tried all the suggestions in previously raised issues regarding this, all to no avail.

Also below are two pictures of a simple bowl I print. It prints perfectly if I disable infill and then use modifiers to introduce upper and lower solid layers. However when prepared without modifiers and 5 upper and lower layers, 3 perimeters and 20% infill the amount of Solid Infill it generates in the walls is shocking I think.

So I wonder if it is possible to introduce a simple check box to prevent solid infill from being generated, be it in the Print Setting Infill menu or as a modifier option.

I'm aware its probably not a simple ask, but for me and as I've read for others, its been a frequently requested option that would prevent some frustration.

Many thanks for looking at this.

[image]
[image]
[image]
[image]
[image]
[image]

neophyl, 2023-03-01:
Could you not find https://github.com/prusa3d/PrusaSlicer/issues/1054 ?  IS there any reason to open yet another duplicate to the VERY long running open issue.

paulluby, 2023-03-01:
> Could you not find #1054 ? IS there any reason to open yet another duplicate to the VERY long running open issue.

Went though that but it seemed to be just a discussion about workarounds?

neophyl, 2023-03-01:
No its not a discussion , it is a request to fix the very long running problem.  Admittedly there have been lots of offered 'workarounds' in that issue but none of them work all the time which is why many would like to see it addressed properly.

While I would love to see it resolved, opening duplicates isn't really going to get it done any sooner.  I'd say at least half the recent issues opened have been duplicates (just see the organic supports go through my model or organic supports start in midair for examples) and its just a waste of resources for the dev's to keep closing them.

bubnikv, 2023-03-01:
We are working on a solution.

paulluby, 2023-03-01:
Thanks for letting me know.

Should I close this or leave it open?

dehmlowm, 2023-03-01:
> We are working on a solution. st 1. 3. 2023 v 12:38 odesílatel neophyl ***@***.***> napsal:
> […](#)

You are a king among men! 

Cross referencing, offer stands:
https://github.com/prusa3d/PrusaSlicer/issues/9245

lukasmatena, 2023-03-06:
This should be fixed in [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5). Feedback is welcome.

k1-801, 2023-03-06:
As a temporary solution, you could disable the Arachne perimeter generator. For some reason, it worked every time when I was trying to solve this.

dehmlowm, 2023-03-06:
> Hi guys.
> 
> Definitely reduces the vibration, but for me still doesn't allow me to stop the solid infill all together.
> 
> In all honesty it appears to have increased the amount of solid infill.
> 
> The reduction in vibration is dramatic I must admit.
> 
> I tend to reduce the amount of solid infill using a Height Range Modifier so I'll have to continue doing that for the pilots but the reduction in vibration has made printing the bowls a much less agonising experience for my printers.
> 
> I've been watching my printer printing a test bowl sliced with Alpha 5 for the last 30 minutes and the difference in vibration level is astounding, as in there isn't any.
> 
> So for me while it would be nice to have a check-box that turned off Solid-Infill totally, the issue with my pilots is exactly what modifiers are for I suppose, so forget that.
> 
> Yet again well done guys, keeps getting better and better.
> 
> [image] [image] [image] [image] [image]

You might play with these settings if you haven't already:
[image]

Bring them down to the the thickness of the the wall:
[image]

mardon1975, 2023-10-14:
Do we have an update on this fix you guys are putting in?

carlosjln, 2024-07-06:
> Do we have an update on this fix you guys are putting in?

@mardon1975 yeah... we're all still waiting :)

k1-801, 2024-07-07:
> > Do we have an update on this fix you guys are putting in?
> 
> @mardon1975 yeah... we're all still waiting :)

+1

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena