prusa3d/PrusaSlicer #9899: Temperature for second layer is emitted at wrong place

NeilW999, 2023-03-01:
### Description of the bug

I would expect the "Other_layers" tempertaure which effects all layers after the first to be emmited before or between custom G-code for layer changes. Interestingly this differs from the behavior of Slic3r which emits the temperature change before both of the custom G-code for layer change.

I have a Revo nozzle and want to always print at a slightly higher temperature for all filaments. It is easy to change the first layer temperature in the custom "Start G-code".  However I can't change the temperature of the second layer because Prusaslicer outputs the "Other layer" temperature after the custom G-code for layer changes. My initial approach was to add the following line to the "After layer change G-code":

;AFTER_LAYER_CHANGE
; Z [layer_z] N [layer_num] H1=[first_layer_height] HN=[layer_height]
{if layer_z > first_layer_height }M104 S{temperature[0]+10} ; set Nozzletemp Add 10 for Revo{endif}

This results in the following output at the second layer change.

[image]

### Project file & How to reproduce

To reproduce. Add custom code to the "After layer custom G-code" and set the "Other layers temperature"  to a different value to the "First layers temperature".

[Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/10861336/Body1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64 and 2.6.0-alpha4+win64

### Operating system

Windows 10

### Printer model

Prusa MK3S+

tjhalva, 2023-03-02:
Might be related to: https://github.com/prusa3d/PrusaSlicer/issues/8911

NeilW999, 2023-03-02:
> Might be related to: #8911

I looked at this before posting this issue and I think this issue is quite different. If "Other layer temperature" is different to first layer then the G-code to change temperature is emmited at the wrong point, which is after both custom G-codes for layer change.  The temperature should be written between or before the custom G-codes for layer change so the user can overide it.

It is not ignoring the temperature as #8911 states. This is about when the temperate G-code is output for the second layer.

closed 2026-09-29 by prusaslicer-bot[bot]