prusa3d/PrusaSlicer #9902: 2.6.0-alpha4: Extra length on restart not working

InstantBenHur, 2023-03-01:
### Description of the bug

If I set a value >0mm for 'extra length on restart', nothing happens. In PrusaSlicer 2.5.0 the setting works.

For example, if I set a value of 10mm in 2.6.0-alpha4:
[image]
Same project, but 2.5.0:
[image]

I know that 10mm don't make sense. It's only that high to make it more visible.

### Project file & How to reproduce

Project file with extra length on restart set to 10mm in printer settings:
[Tomatentrichter2.zip](https://github.com/prusa3d/PrusaSlicer/files/10864664/Tomatentrichter2.zip)

Just set a value >0mm for extra length on restart in printersettings or filament settings.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Windows 11 x64

### Printer model

Prusa i3 MK3S+

rtyr, 2023-03-02:
Reproduced. Thanks.

PascalGodin, 2023-03-12:
Noticed the same on Alpha 5. Was about to open new issue but found yours.

lukasmatena, 2023-11-21:
This should be fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1), it was flushed with the ramping travel implementation. Closing.

closed 2023-11-21 by lukasmatena