prusa3d/PrusaSlicer #9906: [Feature Request] Cloud Sync / Local Sync and Market / Rating System for Prusa Slicer Profiles and Filament Settings

jaddel, 2023-03-02:
I'd like to start by expressing my appreciation for the developers behind Prusa Slicer. Their hard work has created a powerful and user-friendly slicing software that has helped countless 3D printing enthusiasts. I have a feature request that I believe could enhance the Prusa Slicer experience.

Problem: Manually transferring Prusa Slicer profiles, settings, projects, and filament settings between computers is frustrating and error-prone. Sharing settings with others is difficult.

Solution: Add a cloud sync feature to Prusa Slicer where users can sync profiles, settings, projects, and filament settings across instances of the software. Users can discover and share settings through a rating system. 

How it works: Users sign in to their cloud account within Prusa Slicer and select which files to sync. The software automatically syncs files across instances. Users can rate and review settings and share their own.

Alternative: Users could manually specify a cloud-synced folder location for profiles and settings.

Context: Cloud sync improves user experience by making printing consistent, discovering new settings easier, and switching between computers seamless.

Related Requests:
https://github.com/prusa3d/PrusaSlicer/issues/4898
https://github.com/prusa3d/PrusaSlicer/issues/5023

Gregregorovich, 2023-03-02:
As a workaround, you could use a cloud syncing app that supports symbolic links.

OneDrive doesn't natively support this, but you can force it to with OneDrive Bully, as it doesn't detect changes to symbolic links (as the link file hasn't changed), but a manual sync (or automated sync via OneDrive Bully) does detect the change in the file.

Jan-Soustruznik, 2026-01-12:
closed as duplicate of #8891

closed 2026-01-12 by Jan-Soustruznik