prusa3d/PrusaSlicer #9908: FR: Function to compensate for bowden design and delayed extrusion after certain types of travel moves

rasssta, 2023-03-02:
**Is your feature request related to a problem? Please describe.**
I have a problem with Prusa Mini where I get delayed extrusion after a long travel of the x-axis from left to right on the build plate. This  issue seem to be related to the bowden design and the long PTFE tube from the extruder. I have described the problem in-depth in an issue for the firmware repo here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997

**Describe the solution you'd like**
As I assume it would be hard to make a universal fix for this issue, a work around could be to add a new function similar to the "Extra length on restart", but only after print moves from left to right. I have tried to use the "Extra length on restart" function but it creates an overflow of filament on the part of the object where travel is done in the opposite direction (from right to left), so the problem is basically just "moved" when using that function.

**Describe how it would work**
Ultimately it would be nice to have a function that estimates the travel length and positions and takes the bowden design into account to calculate the amount of extra filament to use on restarts to achieve the correct pressure in the nozzle.

**Describe alternatives you've considered**
See above.

**Additional context**
I have added 3 test STLs in the other issue that can be found here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997
It seems like this problem also exist in the Prusa MK series as there's another issue for it here: https://github.com/prusa3d/Prusa-Firmware/issues/2760

ryanklafuric, 2023-04-13:
There's 46 comments on the bug associated with this feature request. Seems like it'd be a worthwhile addition to the slicer. 

If a slicer update isn't possible, it may be worth a hardware update to the Mini to fix it's tight bowden-tube pathing or to replace the bowden setup altogether (e.g. a lightweight direct drive setup eliminating the play that's inherent to bowden extruder designs).

rasssta, 2024-01-06:
Proof of Concept code can be found here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997#issuecomment-1877041956

closed 2026-09-29 by prusaslicer-bot[bot]