prusa3d/PrusaSlicer #9910: Stützmaterial: Organisch / Zwei Objekte

BigBen2003, 2023-03-02:
### Description of the bug

Im Slicer sind zwei Objekte automatisch ziemlich dicht nebeneinander platziert worden. Soweit in Ordnung
Bei aktiviertem Stützmaterial "Organic" werden die Stützen direkt im auszudruckendem Objekt platziert:

[image]

[image]

Wenn die Objekte so ausgedruckt werden, könnte man die Stützmaerialien nicht vollständig entfernen, da diese vom Objekt eingeschlossen sind.

### Project file & How to reproduce

[Abdeckung 2 Fehler.zip](https://github.com/prusa3d/PrusaSlicer/files/10873872/Abdeckung.2.Fehler.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0 alpha3

### Operating system

Windows 10

### Printer model

Prusa MK3S

neophyl, 2023-03-02:
https://github.com/prusa3d/PrusaSlicer/issues/316

kubispe1, 2023-03-08:
Thanks, closing

closed 2023-03-08 by kubispe1