prusa3d/PrusaSlicer #9914: alternate wall thickness - interlocking of wall and infill

jovannix, 2023-03-02:
Cura Slicer has a function called "Alternate Extra Wall". This leads to an extra wall every other layer and therefore to an interlock between infill and wall.
I’m almost sure that this function was available in older Prusa slicer versions: e.g. by typing Wall Thickness „2.5“ (two and a half), the wall alternates 2-3-2-3-2…
Is it possible to implement such function in the Prusa Slicer (again)?

rtyr, 2023-03-02:
Closing as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1011.

closed 2023-03-02 by rtyr

jovannix, 2023-03-06:
you closed the new feature request.
Do you already work on it or is it not possible to implement this feature?

Best regards,
jovannix

rtyr, 2023-03-06:
We are considering it. I closed it because there is already similar request.