prusa3d/PrusaSlicer #9916: Calibration Tools Tab

SeaDoubleU, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
All Printers still require a lot of calibration.  From initial setup to every time you switch filaments. 

**Describe the solution you'd like**
 It would be nice to not have to go to a site, fat finger in a bunch of parameters, and spit out a test, the slicer should be able to to those sorts of things for you.  A Calibration tab would be awesome.  Something with built in calibration tests, whether is is for initial set up of a machine, like E-steps, or Linear Advance when you switch to a new filament.  

**Describe how it would work**
Directly select items from the toolbar under a Calibration heading.  Further integration for automatically loading the g-code when you select the appropriate value(s) for items like linear calibration.

**Describe alternatives you've considered**
links to tests where the slicer uploads the setting in your project into the requested fields

**Additional context**
Add any other context or paint drawing about the feature request here.

JEBHUB, 2024-10-07:
Yes, I agree with this.  Having to download models for calibration or input settings  into a website isn't the best.  Half the time the website settings are sliced in a different software.  It only gets your results close.
 Having a calibration tab in PS would be a great feature!  Please make it happen PS genius's, we love you long time!

feyded1020, 2024-10-19:
How is this still not a thing, Orca has done it correctly, just fork that work into your slicer and modify what is necessary. This is a requirement for a dialed in printer.