prusa3d/PrusaSlicer #9919: More options for Organic Supports

AssiveMass, 2023-03-03:
Hi,

I'd hope for more options for Organic Supports in the coming releases.
Currently there are not enough options to optimize the supports for different objects, I think.

I wish at least the following settings:

**Max height from bed**

A setting that would force the support to attach to the object (if possible) instead of the bed if the maximum distance is exceeded.

[image]

**Max diameter of anchoring point**

Max diameter for saving material when a thick anchor isn't needed.

[image]

**"Avoid supports" modifier**

A modifier that would make the organic support go around the modifier. Would be useful for moving the support further away from the object to get more surface area for the first layer anchor, especially when the main object has a brim. The current support blocker modifier doesn't seem to do this. A setting that would do this automatically for a given distance from object would be nice, but I still feel that a modifier would be useful for some cases, so both please.

Thanks!

ThomasRahm, 2023-03-05:
_"The current support blocker modifier doesn't seem to do this"_
As far as i see they should, but are bugged. It seems you are the only one that noticed, but also did not realize it is a bug.
I am currently trying to change that (there are still some issues though):

[image]

 
[image]

Surrounding a model in support blocker would be a bad idea though, as then there would be no place for the tips to be placed.
If you just want the branches to have a larger distance to the model, the setting you are looking for should be `XY separation between an object and its support`

AssiveMass, 2023-03-06:
@ThomasRahm
Nice, great job.

_"Surrounding a model in support blocker would be a bad idea though, as then there would be no place for the tips to be placed."_

Yeah true. My idea of Support Avoider would still make the supports to contact the model where needed, but avoid the modifier if possible. I guess this gets too complicated for a too niche feature, so fixing the blocker bug would cover 99% of the use cases. The other two settings are more important imo.

neophyl, 2023-03-06:
All the modifiers do is define surfaces on the model where they intersect.  They have never worked as a 'keep out' area and afaik were never supposed to for blockers.  

So Thomas blocker there that intersects with the models base and feet is doing what I would have expected it to do.

ThomasRahm, 2023-03-06:
@neophyl 
I understand that, but nearly all people i have spoken with, would expect tree/organic support branches to avoid the blocker.
In the end i try to focus more on what the user would expect a setting to do, as they are using the software. 
Also it is very useful to avoid branches snaking though hard reachable parts of the model, as one can just place a support blocker in any hole/area a branch is not allowed to go though. 
If one wants a certain overhang not supported i think that paint on support blocker should be used.

In the screenshots above (that are NOT of the current alpha4) the trees avoid the blocker, even though they do not overlap with the base of the model.

@AssiveMass
Regarding your other suggestions: Max height to buildplate would be a lot of work.

The Max branch diameter (called Trunk Diameter in Cura) would be very easy to implement, but it is one of the (in my opinion) less important settings. I can fully understand the need to not have less important settings to keep the UI simple, so that new users are not intimidated by a plethora of settings. If the devs of PrusaSlicer want this to be added, i can open a pull request for the non-UI part of this.

AssiveMass, 2023-03-14:
@ThomasRahm Alright, but I'd still hope for branch diameter settings to be implemented, the default setting wastes material in a lot of cases. Regarding cluttered UI, that's why we have different "difficulty" settings in the slicer, just hide this less important stuff that powerusers would still appreciate under Expert settings, or make a fourth category for even more experienced users. The lack of options is steering the real experts to Superslicer, I think.

ThomasRahm, 2023-03-15:
I am the wrong person to ping though, if someone from PrusaSlicer wants this i am happy to do a pull request. It is a very easy change in the tree support code (I have not looked into how to add the setting to the UI though). 
You should use another name for to describe it though, branch diameter already is a setting if i remember correctly. You want a maximum branch diameter setting iirc.

AssiveMass, 2023-03-22:
One option to replace the "Maximum height from bed" could be a forced snug support for a particular section. You could have an option in the support painter to paint snug instead of organic, with a different key combo and brush color, for example.

[image]
[image]

h0jeZvgoxFepBQ2C, 2023-03-22:
It would be really nice to also add the "branch distance" settting from cura. It really helps, f.e. for my dactyl manuform keyboard it generates way too many small branches:

[image]
[image]

In cura it helps a lot to set the branch distance to min. 6mm.

ThomasRahm, 2023-03-22:
Limit branch reach does only that, limit the reach of each branch. It will never cause less branches!
I think you want to increase the preferred branch angle instead.

h0jeZvgoxFepBQ2C, 2023-03-22:
Hm, I'm not sure (I'm a 3d printing newbie :D).. I only had several prints which had a much better support structure by increasing the "Tree support branch distance"..  See following 2 images from cura:

[image]

Here with "Tree support branch distance" set to 6mm:

[image]

But I guess i dont have any I idea what I'm talking about, just wanted to try the setting out in prusa and saw that it's not available there.

ThomasRahm, 2023-03-22:
Oh that branch distance i misunderstood, i thought you were talking about limit branch reach in the xmas alpha. The corresponding setting in PrusaSlicer should be branch density. A lower branch density results in a higher distance between tips.

h0jeZvgoxFepBQ2C, 2023-03-24:
Hm, I tried to set the density to 30, 60 and 100%, but it nearly looks the same somehow. 
Could you tell me what % means, in Cura its "mm", so one could estimate what comes out.. But not sure what the % mean? 

[image]
[image]
[image]

And thank you so much for your really great alogorithm, its reaaaaaallly cool and helps so much!!! ❤️ ❤️ ❤️ 
Supports looks so much nicer now because of your hard work and efforts 👍 👍 👍 👍

closed 2026-09-29 by prusaslicer-bot[bot]