prusa3d/PrusaSlicer #9920: Continuous ironing

skurt, 2023-03-03:
Ironing is a brilliant feature.
I only see one issue with it. Whenever ironing is stop/paused/split in one place (let's call that X), continues and then either goes back or stars ironing again from X, that always causes imperfections and is visible on a final product.

Obviously the solution would be to go continuously with no pauses - and obviously that is not possible in many scenarios.
However, in case of logos, badges, business cards and similar products - it's not only possible, but it also brings the print closer to perfection - I tried.

Here's an example:
https://1drv.ms/u/s!AuozNE2lK2_tgfQjDxNg9NW7w2mBUQ?e=99siIt
Please don't focus on the top layer, focus on the ironing on the big square.
One of those prints is sliced using Prusaslicer's default settings
The other is the same, except for the ironing in the middle - I provided that.
I won't tell you which one is which, as you'll be able to tell.

I can do this on a square, I would probably struggle to do it for a round. And I definitely wouldn't be able to do it for something irregular.
But you guys - you know the shape of whatever surface you're going to iron and so...if you iron the whole thing with no regard what's going to be in the higher layers (in my example the round bit) - boom, done.

I'd love to somehow clone and fork this thing here or whatever it is that one should do and update it to have this already implemented, but unfortunately I'm too dumb to go that far.
But you guys ain't - look what you've built :) So please at least consider this.

k1-801, 2023-03-03:
Duplicates #9451

skurt, 2023-03-03:
Yes, that is, I believe, exactly what I'm looking for.

rtyr, 2023-03-03:
Ok, thanks. Your issue will be linked there.

k1-801, 2023-03-03:
> That issue wants a smooth surface with no regard what's **under** it (as I understand it)

Actually, no, it's for both above and below, and mostly - for above.