prusa3d/PrusaSlicer #9921: Slicing is stuck when using organic supports and nozzle diameter < 0.2

piontec, 2023-03-03:
### Description of the bug

When I use organic supports and nozzle size >= 0.2, everything works. When I change the nozzle size to 0.15 or 0.1 (I use them sometimes), slicing is stuck on "generating supports" and never (tried at least 15 mins) ends.

### Project file & How to reproduce

1. Slice with nozzle size 0.2 - should work
2. Change nozzle size to 0.15 and reslice - slicer gets stuck.
[Hero_Forge_Explorer_Demo-prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/10882530/Hero_Forge_Explorer_Demo-prusa.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha4+linux-x64-GTK3-202302180033

### Operating system

Ubuntu 22.10

### Printer model

Ender 3 pro

bubnikv, 2023-03-03:
How much RAM do you have in your computer? You may be low on RAM for such object.

piontec, 2023-03-03:
24 GB. And slicing with 0.2 takes maybe 1 min. Also, process is not OOM killed. I also checked mem usage while slicing with 0.2 and 0.15 mm - I have 14 GB free in both cases. Does it slice with 0.15 for you?

kubispe1, 2023-03-14:
SPE-1596

closed 2026-09-29 by prusaslicer-bot[bot]