prusa3d/PrusaSlicer #9922: Cut feature: when cut plane cuts the part into more than 2 pieces, allow to select which ones go to which object

k1-801, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
A good example can be seen on this screenshot (model is SFW):
[image]

Say, I'm trying to cut an arm on this model, to print it separately with a connector (supports typically fail on it's rifle).
No matter how I orient the plane, I have to take into consideration the parts getting cut out of: the helmet, the belt, and the rifle itself.
A similar thing occurs when trying to put a cut so the head can be rotatable on the connector: parts of one or both shoulders get cut off with it.

**Describe the solution you'd like**
The slicer already shows the perimeter of the actual cut that is about to be made. It should be able to check how many objects will actually be cut off by this action, which parts of the half that will stay in place will end up just hanging in the air.
If there is more than one sub-part that will be made by the cut (parts of belt/helmet in the example), the user should be able to opt-out of separating some of them (possibly by clicking on the sub-part). When there is a sub-part that will stay in place and will end up just hanging in the air (side handle of the rifle in the example), the slicer should allow to let this subpart stay on the piece being cut off, or become a third independent object. 

**Describe alternatives you've considered**
No alternatives found with only the slicer itself, CAD software would be required to manually alter the model.

daltonnyx, 2023-03-04:
I have same issue when trying to cut complicated object. May be if we can define the x-y size of the plane, so we can cut only part of the model we need.

k1-801, 2023-03-05:
> May be if we can define the x-y size of the plane, so we can cut only part of the model we need.

Actually, this might induce even more problems.
1) A cut with a size limited rectangularly is not guaranteed to cut through the entire intersection of the plane with the model
2) If it does, it may have positioning issues when trying to arrange the cut off part on the cut plane on the table
3) It is never guaranteed that the problematic pieces can be discarded from being cut by using a limit rectangle, especially axis-aligned one.

The graph component-based approach with simply showing the independent cut perimeters, independent objects and simple click-to-remove interface should be able to solve all this, but might weirdly enough produce performance issues when the plane is located too close to the surface, making every relief feature a separate component.

foreachthing, 2023-03-05:
Cut with cube, cylinder or any other geometric shape comes to mind.
A plane has to be "infinite"... a cube could be viewed as finite plane.

kubispe1, 2023-05-17:
Hi, this improvement has been implemented in PrusaSlicer 2.6.0-beta1. I am closing this issue.
Thanks!

closed 2023-05-17 by kubispe1

seanmcne, 2023-10-25:
> Hi, this improvement has been implemented in PrusaSlicer 2.6.0-beta1. I am closing this issue. Thanks!

I have a really basic question, can you point me to how this was solved?  I was trying to find documentation on it but didn't see anything - any pointers to how to accomplish this would be helpful.  Thanks!

YuSanka, 2023-10-25:
@seanmcne, To select/deselect some parts use a right mouse click. See [change log for PS 2.6.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta1)