prusa3d/PrusaSlicer #9928: clipping plane views for GCODE preview.

yahbluez, 2023-03-04:
It is not a problem but a feature request.

Clipping planes like we have in the cut tool 
to have a preview into the sliced object.

After slicing allow a clipping plane like we have with the cut tool to have any look into the inside of a sliced object.

This feature would be new and unique to prusaslicer.

I'm sure this feature is very easy to implement, it can be done by just cutting a model after slicing and show the resulting image after the cut.

foreachthing, 2023-03-05:
I still use CraftWare for exactly that feature.

yahbluez, 2023-03-06:
Thx for the tip. It is a useful function and having that inside the slicer will enhance the workflow a lot.

Tupson444, 2026-09-14:
still relevant