prusa3d/PrusaSlicer #9931: FR: Ability to switch left/right mouse buttons for camera control

JacobH4, 2023-03-04:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
The ability to rotate the camera around the obect/buildplate while holding the right mouse button instead of the left mouse button.

**Describe how it would work**
A toggle option in the camera preferences to switch the camera rotate feature to the right mouse button.

**Describe alternatives you've considered**
Remembering to use it how it's currently setup.

**Additional context**
Feature could more closely resemble other CAD/design programs where the left mouse click moves an object, right mouse click rotates the camera, and the middle mouse pans the camera.

NoeliaG2018, 2023-05-19:
OMG yes please, I need this. It is extremely annoying-

vgdh, 2023-11-19:
yeeeees

Calibrumm, 2024-10-23:
I know its minor but this, please. all my modeling and CAD software has left mouse select, right mouse rotate, and middle click pan/zoom

closed 2026-09-29 by prusaslicer-bot[bot]