prusa3d/PrusaSlicer #9932: PrusaSlicer 2.6.0-alpha3 crashes on open

mreynolds0404, 2023-03-04:
### Description of the bug

macOS 12.6.3, Intel machine

Application crashes after splash screen, prior to main UI being rendered.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [851]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha3+MacOS-x64-202302031521 (PrusaSlicer-2.6.0-alpha3+MacOS-x64-202302031521)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-03-04 17:03:47.1560 -0500
OS Version:            macOS 12.6.3 (21G419)
Report Version:        12
Anonymous UUID:        D620A3D2-972A-94CC-F8F2-AAA39676E1C6

Time Awake Since Boot: 140 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000014
Exception Codes:       0x0000000000000001, 0x0000000000000014
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [851]

VM Region Info: 0x14 is not in any region.  Bytes before following region: 140737488101356
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7ffffffc2000-7ffffffc3000 [    4K] r-x/r-x SM=ALI  

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x1125460a5 ImFont::CalcTextSizeA(float, float, float, char const*, char const*, char const**) const + 69
1   PrusaSlicer                   	       0x1124df930 ImGui::CalcTextSize(char const*, char const*, bool, float) + 112
2   PrusaSlicer                   	       0x110bcbb64 Slic3r::GUI::NotificationManager::HintNotification::count_spaces() + 36
3   PrusaSlicer                   	       0x110bcd63d Slic3r::GUI::NotificationManager::HintNotification::init() + 29
4   PrusaSlicer                   	       0x110bce599 Slic3r::GUI::NotificationManager::HintNotification::retrieve_data(bool) + 425
5   PrusaSlicer                   	       0x110b9d13d Slic3r::GUI::NotificationManager::HintNotification::HintNotification(Slic3r::GUI::NotificationManager::NotificationData const&, Slic3r::GUI::NotificationManager::NotificationIDProvider&, wxEvtHandler*, bool) + 221
6   PrusaSlicer                   	       0x110b9a026 Slic3r::GUI::NotificationManager::push_hint_notification(bool) + 310
7   PrusaSlicer                   	       0x11087dcf0 Slic3r::GUI::GUI_App::post_init() + 1936
8   PrusaSlicer                   	       0x11089f3cc wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_6>::operator()(wxEvtHandler*, wxEvent&) + 76
9   PrusaSlicer                   	       0x1128ada1e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
10  PrusaSlicer                   	       0x1128ad78b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
11  PrusaSlicer                   	       0x1128ad6a0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
12  PrusaSlicer                   	       0x11280668a wxAppConsoleBase::ProcessIdle() + 74
13  PrusaSlicer                   	       0x11261a8f6 wxAppBase::ProcessIdle() + 22
14  PrusaSlicer                   	       0x11278c77a wxApp::ProcessIdle() + 26
15  PrusaSlicer                   	       0x1128b76ec wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 92
16  CoreFoundation                	    0x7ff81279f520 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
17  CoreFoundation                	    0x7ff81279f3b2 __CFRunLoopDoObservers + 543
18  CoreFoundation                	    0x7ff81279e967 __CFRunLoopRun + 1131
19  CoreFoundation                	    0x7ff81279de3c CFRunLoopRunSpecific + 562
20  HIToolbox                     	    0x7ff81b44d5e6 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	    0x7ff81b44d34a ReceiveNextEventCommon + 594
22  HIToolbox                     	    0x7ff81b44d0e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
23  AppKit                        	    0x7ff8151d7fad _DPSNextEvent + 927
24  AppKit                        	    0x7ff8151d666a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
25  AppKit                        	    0x7ff8151c8d19 -[NSApplication run] + 586
[... 357 lines cut ...]
USB Device: USB30Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

### Project file & How to reproduce

Open application.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

macOS 12.6.3

### Printer model

FLSUN V400, Prusa Mini

ulab, 2023-03-06:
Does this still happen with alpha5?

mreynolds0404, 2023-03-10:
this does not happen in alpha5, but now I have another issue: no window. I just get the menu bar with no other UI components. Importing an STL does not make anything appear.

closed 2026-09-29 by prusaslicer-bot[bot]