prusa3d/PrusaSlicer #9933: support material, non-printable layer raft.

konoja, 2023-03-05:
### Description of the bug

Hello, good afternoon. I am writing for the following, how can I use the raft option in the support material box. since in all the ALPHA versions the following error appears. this seems with support or without these

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

prusa slicer 2.6 alpha1, alpha2, alpha3, alpha4

### Operating system

windows10

### Printer model

all prints wit suports

dorigo077, 2023-03-06:
I was going to report this same issue, its seams cannot use raft and organic support.  It looks like skips the first layer

Raft works fine with grid and snug supports types.

prusa slicer 2.6 alpha4

windows 11

[image]
[image]
[image]
[image]
[image]
[image]

bubnikv, 2023-03-06:
Rafts with organic supports will be fixed in PrusaSlicer 2.6.0-alpha5 to be released soon.

closed 2023-03-06 by bubnikv