prusa3d/PrusaSlicer #9934: Access Violation

Krueger4Runner, 2023-03-05:
### Description of the bug

I tried to slice an object with text encoded onto object. I got an Access Violation error. How do I get past this?
[image]

### Project file & How to reproduce

[Creep Cast ID Colorized 1.zip](https://github.com/prusa3d/PrusaSlicer/files/10890804/Creep.Cast.ID.Colorized.1.zip)
I added text and deleted the text and it still has a Access Violation Issue, to recreate you just need to slice it

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3+win64-202302031521

### Operating system

Window

### Printer model

Prusa MK3S & MK3S+ MMU2S

lukasmatena, 2023-03-05:
I was not able to reproduce this with any of the 2.6.0 alphas.

kubispe1, 2023-03-08:
@Krueger4Runner Please can you check with 2.6.0-alpha5 and let us know?

lukasmatena, 2023-03-08:
Thanks. Closing.

closed 2023-03-08 by lukasmatena